AndrewMascolo/OnewireKeypad #1: 3v3 configuration

oldmanegan, 2015-11-25:
Your note on the Arduino Playground web page states: 
- Vref = +5V. (Please note, 3.3V is not stable enough to produce accurate analog readings)

I tried this with a 3v3 32U and set the resistors as follows:
4700 converted to 2400
1000 converted to 680
Tested OK with a 4x4 Keypad.

Works consistently. You may wish to update your notes and the Playground and any Readme you may have here

vanwykc, 2017-12-08:
The difference in the ADC code between two buttons determines the noise immunity of the configuration.  I've done some quick calculations and found that the noise immunity is improved about 4 times if you make value of the resistor to ground 10x that of the column resistors (The 680 Ohm resistors in your example).  This also makes the keypad much less sensitive to the tolerance of the resistor values.

(4x better because smallest difference in an 8-bit ADC's result = 1 if all bottom resistors = 680 Ohm, but when the single 680 Ohm resistor between the sense pin and GND is replaced by a 6800 Ohm one, then the smallest difference in an 8-bit ADC's result = 4)

AndrewMascolo, 2017-12-08:
Thank you. 

The values I’m using are just what I had available when I made my keypad. You can use whatever you want to use, the resistor calculator should be able to handle it. 

As a future addition, I was thinking of tweaking the code a bit to also handle diodes too.

- Andrew