AndrewMascolo/OnewireKeypad #4: Inaccuracy due to low voltage

zveras, 2016-11-05:
My arduino connected to USB provides only ~4.8v current, that makes troubles reading last row ( * and 0 returned 0, # and D returned D). I used 4.7k and 1k resistors like in your example. I solved problem changing resistor near output to 5.7k, but your code does not allow that change. I have updated your code adding R3. 

   OnewireKeypad( T &port, char KP[], uint8_t Rows, uint8_t Cols, uint8_t Pin, int R1, int R2 , int R3, long Precision)
      : port_( port ), latchedKey( BitBool< MAX_KEYS >() ), _Data( KP ), _Rows( Rows ), _Cols( Cols ), _Pin( Pin ), holdTime( 500 ), startTime( 0 ), lastState( 0 ), lastRead( 0 ), ANALOG_FACTOR(1024 / 5.0), Num( 0 ), R1( R1 ), R2( R2 ), R3 ( R3 ),  Precision( Precision ) { }

and this part:

      float V = (5.0f * float( R3 )) / (float(R3) + (float(R1) * float(R)) + (float(R2) * float(C)));

AndrewMascolo, 2016-11-05:
That would work, but now you're adding another resistor to the calculations. What you should have done was just change the 5.0f to 4.8f and also the 5.0 in ANALOG_FACTOR(1024 / **5.0**) to 4.8.

I updated the library that will allow you to tell the code what voltage to use in the calculations.
**SetKeypadVoltage( 5.0 )**
Add this to your setup() and just change its value to 4.8 or to whatever voltage your keypad is getting from the Arduino.

zveras, 2016-11-05:
Changeing voltage does not help, 4,7k resistors are too larga and currency output is very low and unstable. I am not adding additional resistor, I am adding additional variable. 
Vout = Rgrnd / ( Rkeys + Rgrnd)  * Vin  --- in your calculations Rkeys = Rcol + Rrow and Rgrnd == Rcol. You can use other resistor near the ground, not only with Rcol or Rrow value. 

with vin 5v, and 4,7k row, 1k col, 1k grnd resistors:
Row 4 voltage: 0,33 0,31 0,29	0,27

with vin 4,8v, and 4,7k row, 1k col, 1k grnd resistors:
Row 4 voltage: 0,318 0,298 0,28 0,265

with vin 4,8v, and 4,7k row, 1k col, 3,3k grnd resistors:
Row 4 voltage: 0,91 0,86 0,816 0,776

As you see, changing input voltage 5->4.8 parameter does not solve anything (0.1v difference between keys, what makes some keys unreadable due to inaccuracy), unless you use 3.3v and 
different resistors (like oldmanegan recommended). Changing Rgrnd 1k -> 3.3k changes voltage difference by 0.5v between keys

I am not guru at electronics, but only changing resistor near ground pin solved my problem. I am unable to change Rcol and Rrow, because I have already soldered them.

I can upload you xls with calculations.

AndrewMascolo, 2016-11-05:
If that works for you, then by all means, use it. 

I will say this though, your issue does intrigue me as my keypad is only getting 4.3 volts yet it is able to show all the correct keys when pressed. I'll see if there are any ways to get this to work at any voltage with the current resistors. There is always a way.

kokroff, 2017-03-25:
Hello, 
At first, thank you both for very helpfull comments and for a hard work.
The modification by zveras works, you just need to make a change in the library code, don't forget to add R3 here: "long R1, R2, R3;", otherwise you can't compile it and also add the new input to the sketch:OnewireKeypad <Print, 16 > KP2(Serial, KEYS, 4, 4, A0, 4700, 1000,3000, ExtremePrec ); if your GND resistor is 3k.

Andrew

AndrewMascolo, 2017-03-26:
Changes made