AndrewMascolo/OnewireKeypad #7: Analog Keypad wrong ranges

dragoblaztr, 2017-08-30:
I have this keypad, it's 3x4 with analog output and it has 3k resistors for rows and 1k resistors for columns but the pulldown resistor is 10k instead of 1k as in the schematic you provide, the range of analog readings are totally diferent due this issue.

My analog values are

button 1 = 1023
button 2 = 930
button 3 = 850
button 4 = 790
button 5 = 730
button 6 = 680
button 7 = 640
button 8 = 600
button 9 = 570
button * = 540
button 0 = 510
button # = 490

How to fix this

because the range given by your default and changing resistor values is this

OnewireKeypad <Print, 12 > KP(Serial, KEYS, 4, 3, A0, 3000, 1000, MediumPrec );

V:5.00, AR: 1023.00 | V:2.50, AR: 511.50 | V:1.67, AR: 341.00 | 
--------------------------------------------------------------------------------
V:1.25, AR: 255.75 | V:1.00, AR: 204.60 | V:0.83, AR: 170.50 | 
--------------------------------------------------------------------------------
V:0.71, AR: 146.14 | V:0.63, AR: 127.87 | V:0.56, AR: 113.67 | 
--------------------------------------------------------------------------------
V:0.50, AR: 102.30 | V:0.45, AR: 93.00 | V:0.42, AR: 85.25 | 
--------------------------------------------------------------------------------

[image]
[image]
Regards

dragoblaztr, 2017-08-30:
Solved, I look into the h file and find the constructor can add a third resistor value to do that, so I note that I tried the example and thorws a few wrong keys by bad debouncing mostly # after some keys

How to fix this

AndrewMascolo, 2017-08-30:
The code assumes the pull down resistor is the same as the other column resistors (1k). 

I would need to add a way to change this. Give me a few days and I should have a fix for you.

dragoblaztr, 2017-08-30:
@AndrewMascolo 
I already fix this as next

OnewireKeypad <Print, 12 > mk1(Serial, KEYS, 4, 3, A0, 3000, 1000, 10000, 30 );

But sometimes it throws inacurrate values (I think poor debouncing)

Thank you in advance

dragoblaztr, 2017-08-30:
In the function just replaced

`float V = (voltage * float( R2 )) / (float(R2) + (float(R1) * float(R)) + (float(R2) * float(C)));`

with this

```
float V = 0;
      if (R3 != 0) {
        V = (voltage * float( R3 )) / (float(R3) + (float(R1) * float(R)) + (float(R2) * float(C)));
      } else {

        V = (voltage * float( R2 )) / (float(R2) + (float(R1) * float(R)) + (float(R2) * float(C)));
      }
```

AndrewMascolo, 2017-09-01:
The denounce is based on the precision number, so instead of 30, try 40 or 50