AndrewMascolo/OnewireKeypad #11: To get started you need add some data into the constructor

OslafEman, 2018-10-01:
Hi, I'm new to Arduino and there's something I do not understand with this library. I do not understand when you say "you need add some data to the constructor" and it would help me a lot if you could explain to me which is the constructor and maybe how to add that data.
This is surely a silly question and I apologize for that.

AndrewMascolo, 2018-10-01:
Using my example: OnewireKeypad_Final

This is the constructor:
OnewireKeypad <Print, 16 > KP2(Serial, KEYS, 4, 4, A0, 4700, 1000 );

“OnewireKeypad <Print, 16 > ” is the keypad library, this is also referred to as the Type.

“Print” is the library for what you are using to see the characters. Print is the library for the Serial Monitor. 

“16” are the total number of keys your keypad has. In this case the constructor above is using a 4 by 4 keypad.

“KP2” is what I chose to call my constructor reference. You will see it being used further in the code.

“Serial” is the Serial instance being passed in to tell the code to output the key presses to the Serial monitor.

“KEYS” is my array of characters I want to show when that particular key is pressed.

“4, 4” Are the number of rows and columns the keypad has. Ie. 4 rows and 4 columns

“A0” is the pin you are using to read the voltage signals from the keypad.

“4700” is the resistor value you are using for the rows

“1000” is the resistor value for the columns + the pull down resistor.

Hope this helps.

OslafEman, 2018-10-02:
Hello Andrew, I am very grateful for your kindness in answering me.
I have made the suggested changes to your example OnewireKepad_Final, however there is no output for the serial monitor (code and monitor at 115200 baud) and when I press any key nothing happens.
When I upload your ShowRange example I get this output.
[image]
The code of OnewireKeypad_Final I have it like this:
`// Creator: Andrew Mascolo

#include <OnewireKeypad.h>

char KEYS[] = {
  '1', '2', '3',
  '4', '5', '6',
  '7', '8', '9',
  '*', '0', '#',
};

OnewireKeypad <Print, 12 > KP2(Serial, KEYS, 4, 3, A0, 4700, 1000 );

void setup () {
  Serial.begin(115200);
  KP2.SetKeypadVoltage(5.0);
}

void loop() {
  if ( char key = KP2.Getkey() ) {
    Serial << "Key: " << key << " State: ";
    switch (KP2.Key_State()) {
      case PRESSED:
        Serial.println("PRESSED");
        break;

      case RELEASED:
        Serial.println("RELEASED");
        break;

      case HELD:
        Serial.println("HOLDING");
        break;
    }
  }
}

`
I do not know why it does not work for me. I hope you can help me.
Oslaf

AndrewMascolo, 2018-10-02:
What IDE version are you using? You are the second person to say it doesn’t work. Maybe something broke in the newer version?

AndrewMascolo, 2018-10-02:
My sincerest apologies, there was a problem with the debounce code, it is fixed now.  
please redownload the library or replace the OnewireKeypad.h file with the latest one.

OslafEman, 2018-10-03:
Dear Andrew, thank you very much for your kind help. I followed your instructions downloading and installing the files again. I'm using the last IDE version.  Unfortunately I have not been able to operate the keyboard with the library. Of the four examples attached, only ShowRange and KP_Password work for me and return this data.
[image]
[image]

I try to do a project to control the use of coin laundry machines with the Wemos D1 mini platform and I need each pin of it, that's why I thought it was great to use only the analog pin for the keyboard.

OslafEman, 2018-10-03:
Dear Andrew, please find the picture.
[image]

AndrewMascolo, 2018-10-03:
Yea based on your last images, the voltage to resistor range should not have zeros for all the AR values. It should look like this.

I think the resistor values you entered in the constructor are wrong or backwards.
It’s also possible something is wired incorrectly. Your picture of your wiring is kinda blurry, so it is difficult to see the resistors values.

- Andrew

OslafEman, 2018-10-03:
Hello again, the wiring can be seen better in this photo.
[image]

The red wire is + 5V, black is GND and yellow goes to A0, the 3 resistors on the left are 1K and the three on the right are 4.7K (all 1 / 2W and + -5% tolerance) . I connected as best I could by following his illustration on the library's website.
With reference to the constructor, I think it's better if I copy and paste it here, so you can see it.
`// Creator: Andrew Mascolo
//
//#include <Wire.h>
//#include <LiquidCrystal_I2C.h>
#include <OnewireKeypad.h>

#define Rows 4
#define Cols 3
#define Pin A0
#define Row_Res 4700
#define Col_Res 1000
// ExtremePrec, HighPrec, MediumPrec, LowPrec
#define Precision ExtremePrec 

char KEYS[] = {
  '1', '2', '3',
  '4', '5', '6',
  '7', '8', '9',
  '*', '0', '#',
};

//LiquidCrystal_I2C Lcd(0x20,20,4);
//OnewireKeypad <LiquidCrystal_I2C, 12> KP( Lcd, KEYS, 4, 3, A0, 4700, 1000 );

OnewireKeypad <Print, 12 > KP2(Serial, KEYS, 4, 3, A0, 4700, 1000 );
`
Oslaf

AndrewMascolo, 2018-10-03:
Ok everything looks correct, I just don’t know why you are not getting an output. 

Do you have any potentiometers? If so you can use it to debug the problem. 

(Gnd) -vvvvv- (+5)
                 l
             (A0)

- Andrew