AndrewMascolo/OnewireKeypad #12: Serial monitor

ite32, 2019-01-19:
hi,
I have connected my 4x4 matrix to an arduino nano but I am not having much look. This is what I can see in the serial monitor. Any help would be appreciated!
[image]

AndrewMascolo, 2019-01-19:
What does it show if you set the keypad voltage to 5.0?

- Andrew

ite32, 2019-01-19:
whenever I use the keypad with 6.0v I get something like: "x x" or "xf" or "xx f"

AndrewMascolo, 2019-01-20:
It sounds like the baud rate is bad but the code and serial monitor match. I’m not really sure what is going on. If you use a different baud rate like 9600, do you see the same results? 

- Andrew