AndrewMascolo/OnewireKeypad #13: Resistors other than 1k and 4.7k combination does not work accurately

tuapui, 2019-02-14:
Sorry i'm pretty new to Arduino and am quite a beginner at coding, so I had plenty of trouble with this. I assumed you could use other values of resistors since the guide i read on Arduino Playground suggested that could be done.

After playing around with other resistance values on 16 key 4x4 keypad because I realize I didn't have 4.7k resistors on hand, I realized 10k + 47k does not work even after changing the values in the sketch,
1k + 2.22k will not read the 4th column accurately. (1st - 3rd column is ok) 220ohm + 330ohm also have accuracy problems.

Please let me know if the problem I found is the fault of my inexperience, or did I really come upon a bug in the calculation?

P.S. my math is horrible and my fundamentals of circuits is horrible so I couldn't really tell what the calculation is about in the library

AndrewMascolo, 2019-02-14:
I added a function to test your resistor values and if the results are not in sequence or are too close to each other, the resistors will not work.

tuapui, 2019-02-14:
ah, i see. just discovered the ShowRange(); function... it was not in your documentation, sorry I missed it. this is handy!

tuapui, 2019-02-14:
ok with the help of the function, i found that 1k on rows and 220 on cols can work, and also that the keypad I am using has a short on the pin sometimes, making column 4 wonky. thanks for the help, and sorry for the trouble!

closed 2019-02-14 by tuapui