AndrewMascolo/OnewireKeypad #14: Not displaying anything

krinton, 2019-04-08:
Hi, I hooked up the connection as it was suggested and nothing is displaying on the Serial Monitor. I am using Arduino 1.8.7 IDE. I have attached the picture below.
[image]

And i used this code from OneWireKeypad_Final from the file.
#include <OnewireKeypad.h>

char KEYS[] = {
  '1', '2', '3', 'A',
  '4', '5', '6', 'B',
  '7', '8', '9', 'C',
  '*', '0', '#', 'D'
};

OnewireKeypad <Print, 16 > KP2(Serial, KEYS, 4, 4, A1, 4700, 1000 );

void setup () {
  Serial.begin(9600);
  KP2.SetKeypadVoltage(5.0);
}

void loop() {
  //  if( KP.Getkey() )
  //  {
  //    Lcd.clear();
  //    Lcd.home();
  //    Lcd.print( F( "KP: " ) );
  //    Lcd.setCursor(4,0);
  //    KP.LatchKey();
  //  }

  if ( char key = KP2.Getkey() ) {
    Serial << "Key: " << key << " State: ";
    switch (KP2.Key_State()) {
      case PRESSED:
        Serial.println("PRESSED");
        break;

      case RELEASED:
        Serial.println("RELEASED");
        break;

      case HELD:
        Serial.println("HOLDING");
        break;
    }
  }
}

Honestly speaking none of the code displays anything on the scree. Please Suggest. Thanks.

krinton, 2019-04-08:
I added KP2.SetAnalogPinRange(1023) in the setup function, it displays but i can press few numbers only and no matter what I press it displays 'D' only.

[image]

AndrewMascolo, 2019-04-08:
Make sure your serial monitor BAUD rate matches what you set it to, 9600.

Remove this line:
KP2.SetKeypadVoltage(5.0);

- Andrew

Sent from my iPhone

> On Apr 8, 2019, at 1:16 PM, krinton <notifications@github.com> wrote:
> 
> KP2.SetKeypadVoltage(5.0);

AndrewMascolo, 2019-04-08:
You need to use either both setKeypad voltage and pin or neither of them.

Double check your wiring too.

- Andrew

krinton, 2019-04-11:
Still the same issue and result. It displays only key pressed as D. 
[image]

AndrewMascolo, 2019-04-11:
There is a function called ShowRange() see what values it returns. If you see all zeros for V or AR then you have something setup incorrectly. Possibly the keypad it self is wired incorrectly.

- Andrew

krinton, 2019-04-11:
Thanks. Funny thing is it does not display as V and VR as I have seen with other people posting in other issues. Mine shows 
[image]
Probably i am doing something wrong and when I run Password function it shows password.h file is missing. Which i couldnot find the zip folder either.

AndrewMascolo, 2019-04-11:
Password.h is not mine, that is a separate class you need to download.

When was the last time you updated the OneWireKeypadLibrary?

- Andrew

AndrewMascolo, 2019-04-11:
I see, you are using the ShowRange Example sketch. 

Don’t use that, there is a function in the OWKP class called ShowRange, please use that.

- Andrew