AndrewMascolo/OnewireKeypad #17: Can't get the numbers to display when numbers are pressed.

jvierbuchen, 2021-04-01:
Andrew,

I am doing a online class with Peter Dalmaris and he uses this Library for one of his sections. At the time of his class he used the MediumPrec statement but when I try that in my sketch the Arduino says 'MediumPrec' was not declared. I did some research on your latest version of the library and it appears that you no longer require that statement in the constructor. I tried running the sketch without it and the sketch will compile. I get all the text to appear on the LCD screen that is in the setup but nothing appears on the screen when I press the buttons. 

[Keypad_I2C.zip](https://github.com/AndrewMascolo/OnewireKeypad/files/6246754/Keypad_I2C.zip)

AndrewMascolo, 2021-04-01:
Another user a while back made some changes to the library. I just merged them in, please redownload the library.

AndrewMascolo, 2021-04-02:
I found the issue, there was an IF statement I missed in SetKeyPadVoltage. 
Either remove SetKeyPadVoltage from your sketch or download my fix.

jvierbuchen, 2021-04-02:
Thank you so much for your quick response. That solved my problem.

closed 2021-04-02 by AndrewMascolo